Xilinx/finn #1154: IP definition not found for VLNV

maires1306, 2024-08-08:
Hi, everyone!

I'm trying to build finn_examples network resnet50_w1a2 but I'm getting:
`ERROR: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:Pool_hls_1:1.0`

As shown in the build process:

[image]

The only parameter I change within /resetnet50/build.py is the vitis platform to "xilinx_u250_gen3x16_xdma_4_1_202210_1", which is the one I have installed.
[image]

Vivado/Vitis 2022.2 and Ubuntu 22.04.4 LTS are being used and the environment variables are set as:
```
export FINN_XILINX_VERSION=2022.2
export FINN_XILINX_PATH=/tools/Xilinx
export VIVADO_PATH=/tools/Xilinx/Vivado/2022.2
export VITIS_PATH=/tools/Xilinx/Vitis/2022.2
export PLATFORM_REPO_PATHS=/opt/xilinx/platforms
export XILINX_XRT=/opt/xilinx/xrt
```

Checking vivado.log, the last error call is 
```
# create_bd_cell -type ip -vlnv xilinx.com:hls:Pool_hls_1:1.0 StreamingDataflowPartition_7_Pool_hls_0
ERROR: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:Pool_hls_1:1.0 
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv xilinx.com:hls:Pool_hls_1:1.0 StreamingDataflowPartition_7_Pool_hls_0"
    (file "make_project.tcl" line 24)
INFO: [Common 17-206] Exiting Vivado at Thu Aug  8 09:06:12 2024...
```
I thought it would be simple as the bnn-pynq example, which run fine without any modifications but apparently I'm still missing something, I just can't understand what it is.

Another point is the build time: from the first log line to the one that returned a problem it took almost 7 hours, is it normal?
```
[2024-08-08 02:10:21,433] Running step: step_resnet50_tidy [1/15]
...
[2024-08-08 09:05:57,094] /home/maires/Documents/mestrado/finn-examples/build/finn/src/finn/transformation/fpgadataflow/hlssynth_ip.py:74: UserWarning: Using pre-existing IP for StreamingDataflowPartition_7_Pool_hls_0
[2024-08-08 09:05:57,095]   warnings.warn("Using pre-existing IP for %s" % node.name)
```

Thanks in advance!

auphelia, 2024-08-16:
HI @maires1306 , 
Can you please try using the new release and see if your issue is resolved?

In general, to debug an error message like this, you can have a look into the folders that are generated by FINN. E.g. from the screenshots it looks like the ip generation for one of the Pool layers was unsuccessful. You can have a look at the `code_gen_ip_gen_...` that belongs to that node and check the vitis_hls.log to get more insights on what could have gone wrong. The folder path gets saved as a node attribute, so you can open the last checkpoint `.onnx` file with netron, navigate to the problematic node and when you click on it, you should see the ip gen path in the list of node attributes.

bwintermann, 2024-08-20:
@maires1306 Additionally I am curious for the last point you raised about the build time. I am currently working on an older version of FINN also on the resnet50_w1a2, with the first step taking around 30s. Could you maybe link which version of the resnet example you are using exactly and maybe print-debug on which transformation in the tidy up step it hangs for 7 hours?