Xilinx/finn #1165: "Only a single StreamingDataflowPartition supported."

ChristiaanBoe, 2024-08-21:
[PFBFINNprobems.zip](https://github.com/user-attachments/files/16689219/PFBFINNprobems.zip)

Hello everyone :), I run into the following error with running my model through the automated buildflow.

```
Building dataflow accelerator from ./PFB.onnx
Intermediate outputs will be generated in /home/mengfei/finn/finn0.10.1/Results
Final outputs will be generated in output_final
Build log is at output_final/build_dataflow.log
Running step: step_qonnx_to_finn [1/19]
Running step: step_tidy_up [2/19]
Running step: step_streamline [3/19]
Traceback (most recent call last):
  File "/home/mengfei/finn/finn0.10.1/src/finn/builder/build_dataflow.py", line 158, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/home/mengfei/finn/finn0.10.1/src/finn/builder/build_dataflow_steps.py", line 378, in step_create_dataflow_partition
    assert len(sdp_nodes) == 1, "Only a single StreamingDataflowPartition supported."
AssertionError: Only a single StreamingDataflowPartition supported.
Running step: step_convert_to_hw [4/19]
Running step: step_create_dataflow_partition [5/19]

> /home/mengfei/finn/finn0.10.1/src/finn/builder/build_dataflow_steps.py(378)step_create_dataflow_partition()
    376     )
    377     sdp_nodes = parent_model.get_nodes_by_op_type("StreamingDataflowPartition")
--> 378     assert len(sdp_nodes) == 1, "Only a single StreamingDataflowPartition supported."
    379     sdp_node = sdp_nodes[0]
    380     sdp_node = getCustomOp(sdp_node)
```

You can recreate this issue by running the attached jupyter notebook and generating the attached ONNX file.  Which I then run through the automated workflow as follows:
```
import finn.builder.build_dataflow as build
import finn.builder.build_dataflow_config as build_cfg
import os
import shutil

model_file = "PFB.onnx"

final_output_dir = "output_final"

#Delete previous run results if exist
if os.path.exists(final_output_dir):
    shutil.rmtree(final_output_dir)
    print("Previous run results deleted!")

cfg = build.DataflowBuildConfig(
    output_dir          = final_output_dir,
    mvau_wwidth_max     = 80,
    target_fps          = 1000000,
    synth_clk_period_ns = 10.0,
    board               = "Pynq-Z1",
    shell_flow_type     = build_cfg.ShellFlowType.VIVADO_ZYNQ,
    generate_outputs=[
        build_cfg.DataflowOutputType.BITFILE,
        build_cfg.DataflowOutputType.PYNQ_DRIVER,
        build_cfg.DataflowOutputType.DEPLOYMENT_PACKAGE,
    ]
)
```
Hope you guys can see what I am doing wrong with my Brevitas code, its only 2 convolutions big so I am surprised I managed to get an error.

Best regards,

closed 2024-08-27 by ChristiaanBoe

simpleintel, 2025-02-25:
I have the same issue when I use kernel size greater than 1. 

conv1 = QuantConv1d(
            in_channels=2,
            out_channels=2,
            kernel_size=3,
            padding=1,
            weight_bit_width=8,
            weight_quant=Int8WeightPerTensorFloat,
        )

ChristiaanBoe, 2025-02-28:
Dear @simpleintel 

My fix for this solution was to create a custom transform flow as 1d convolutions are not supported in the native automated buildflow.

```
model = ModelWrapper(model_streamlined)
model = model.transform(to_hw.InferBinaryMatrixVectorActivation())
model = model.transform(to_hw.InferQuantizedMatrixVectorActivation())
model = model.transform(to_hw.InferVectorVectorActivation())
# TopK to LabelSelect
model = model.transform(to_hw.InferLabelSelectLayer())
# input quantization (if any) to standalone thresholding
model = model.transform(to_hw.InferThresholdingLayer())
model = model.transform(to_hw.InferConvInpGen())
model = model.transform(to_hw.InferStreamingMaxPool())
# get rid of Reshape(-1, 1) operation between hw nodes
```