Xilinx/finn #1191: Quicktest fail because of deprecated api call

kalahel, 2024-09-19:
Hello,

Starting from a fresh install and running  `./run-docker.sh quicktest` I have thoses error : 
```shell
[...]
 => => writing image sha256:b01e605eccedf1632bb74cdc2e8e74fd5bfe35a84f884a11ef42c597d68a246b                                        0.0s
 => => naming to docker.io/xilinx/finn:v0.10.1-dirty.xrt_202220.2.14.354_22.04-amd64-xrt                                            0.0s

 1 warning found (use docker --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 144)

What's next:
    View a summary of image vulnerabilities and recommendations → docker scout quickview 
Obtaining file:///home/mathieu/Tools/finn/deps/qonnx
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [44 lines of output]
      /usr/local/lib/python3.10/dist-packages/setuptools/__init__.py:80: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
[...]
```

I have a lot of similar errors (deprecated api call).

After those messages, all the tests fail because they don't find qonnx, brevitas nor finn.

I tried forcing the installation by changing the dockerfile and adding :
```dockerfile
RUN pip install qonnx
RUN pip install brevitas
RUN pip install finn-base
``` 

Now I have this test report : `7 failed, 8 passed, 19 warnings, 88 errors`.

It seems that the script can't find some find API : 
```shell
[...]
E   ModuleNotFoundError: No module named 'finn.transformation.qonnx'
[...]
E   ModuleNotFoundError: No module named 'finn.builder'
[...]
```

Did I do something wrong in my installation or is there a deprecation in the docker image ?

Thanks in advance !

auphelia, 2024-09-19:
Hi @kalahel ,
Are you working with current main or dev? 
This error looks like an older version of FINN.

kalahel, 2024-09-19:
Hi @auphelia, thanks for the response !

I followed the [getting started guide](https://finn.readthedocs.io/en/latest/getting_started.html) so just cloned the repo with `git clone https://github.com/Xilinx/finn/`, so I guess I have the current main branch.

auphelia, 2024-09-25:
That is unusual, can you share the exact commands you run in the command line and which env vars you have set?

kalahel, 2024-09-25:
I run `./run-docker.sh quicktest`

My env variable are : 
```shell
export FINN_XILINX_PATH=/home/mathieu/Tools/Xilinx
export FINN_XILINX_VERSION=2024.1
```

auphelia, 2024-09-25:
We are recommending using Vivado/Vitis 2022.2 with FINN currently, but this is not the reason for the issue. 
It might be that you are building from cache and that causes issues. So, you can try rebuilding the docker container. Can you please add `--no-cache` to the run-docker.sh script in the line docker build? To this line: https://github.com/Xilinx/finn/blob/main/run-docker.sh#L198
So: `docker build -f docker/Dockerfile.finn --build-arg XRT_DEB_VERSION=$XRT_DEB_VERSION --tag=$FINN_DOCKER_TAG $FINN_DOCKER_BUILD_EXTRA --no-cache .`
And try to run it again.

kalahel, 2024-09-25:
Unfortunately it did not change the result.
I linked the complete trace if it can help you find clues on what's happening.
[completeTrace.txt](https://github.com/user-attachments/files/17130578/completeTrace.txt)

kalahel, 2024-09-25:
I deleted `deps`, ran `fetch-repos.sh` and re-ran `./run-docker.sh quicktest`, but still the same results.
Here the traces : 
[traceFetchRepo.txt](https://github.com/user-attachments/files/17131567/traceFetchRepo.txt)
[completeTrace2.txt](https://github.com/user-attachments/files/17131570/completeTrace2.txt)

auphelia, 2024-09-26:
Hi @kalahel,
Your error looks like a setup issue. Can you please clarify how exactly your setup looks like?
- Do you have a linux machine?
- Do you have a docker installation?
- What are the exact steps you run in the command line?
- E.g. 
`git clone https://github.com/Xilinx/finn.git` 
`export FINN_XILINX_PATH=/home/mathieu/Tools/Xilinx`
`export FINN_XILINX_VERSION=2024.1`
`bash run-docker.sh`

kalahel, 2024-09-26:
Hi @auphelia,

I'm currently using an Ubuntu 24.04.1 LTS with docker (version 27.2.1) desktop installation.
This is the commands I run from reboot :
```shell
sudo sysctl -w kernel.apparmor_restrict_unprivileged_userns=0
systemctl --user start docker-desktop
export FINN_XILINX_PATH=/home/mathieu/Tools/Xilinx
export FINN_XILINX_VERSION=2024.1
./run-docker.sh quicktest
```

Thank you for your support !

auphelia, 2024-10-11:
Hi @kalahel ,
I was not able to recreate the issue you are/were experiencing. I saw that you commented on other topics in the FINN repos and was wondering if you were able to solve this by yourself. If yes, could you share your solution?

kalahel, 2024-10-11:
Hi @auphelia,

Unfortunatly I was not able to find a direct solution, but now I am using FINN in a WSL2 from Windows, I followed the direction given in this [discussion](https://github.com/Xilinx/finn/discussions/1084).

AliMirhosseini, 2025-01-05:
> Hi @kalahel , I was not able to recreate the issue you are/were experiencing. I saw that you commented on other topics in the FINN repos and was wondering if you were able to solve this by yourself. If yes, could you share your solution?

I am facing this issue as well, and I think it might be due to a mismatch between the test script and changes in the newer version of the repository. The errors indicate, for example, that:
import finn.core.onnx_exec as oxe
results in ModuleNotFoundError: No module named 'finn'.
It seems that the core module has been moved to src/finn/core, and the finn folder is no longer in the main directory. I checked version 0.1 of the repository, and the finn folder was indeed present in the main directory at that time.
I’m not entirely sure if this is the root cause of the issue, so it would be great if someone could review the test scripts to see if they are causing the problem.

Thanks!

AliMirhosseini, 2025-01-05:
> I am facing this issue as well, and I think it might be due to a mismatch between the test script and changes in the newer version of the repository. The errors indicate, for example, that: import [...]

I’ve also tried using FINN in a virtual environment, but the same errors occur.

Hermanstray, 2025-03-12:
I realized this is pretty much what occurs in my issue  #1295 also. I get a bunch of module not found errors from packages that exist in my deps folder. I have tried the things you mentioned @auphelia, with first Ubuntu 24.04 and then with 22.02 and the same error occurs. Has anyone found a solution except from switching to WSL2?