Xilinx/finn #1215: ERROR: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:StreamingMaxPool_hls_0:1.0

IsmailAM1999, 2024-10-16:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [ x] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [ ] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x ] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
Hello,

I encountered this error in dev-branch while trying to Run step: step_set_fifo_depths [13/19]

## Details
I searched the error in vivado.log of the project, and here is the logs :

```
`ERROR: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:StreamingMaxPool_hls_0:1.0
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv xilinx.com:hls:StreamingMaxPool_hls_0:1.0 StreamingMaxPool_hls_0"`
```
I looked at the logs in code_gen_ipgen_StreamingMaxPool_hls_0_xpewuftl folder : 

```
WARNING: [Common 17-1221] Tcl app 'designutils' is out of date for this release. Please run tclapp::reset_tclstore and reinstall the app.
source run_ippack.tcl -notrace
INFO: calling package_hls_ip ip_types=vitis sysgen json_file=/tmp/finn_dev_i22amess/code_gen_ipgen_StreamingMaxPool_hls_0_xpewuftl/project_St>
INFO: Copied 1 ipmisc file(s) to /tmp/finn_dev_i22amess/code_gen_ipgen_StreamingMaxPool_hls_0_xpewuftl/project_StreamingMaxPool_hls_0/sol1/im>
INFO: Copied 8 verilog file(s) to /tmp/finn_dev_i22amess/code_gen_ipgen_StreamingMaxPool_hls_0_xpewuftl/project_StreamingMaxPool_hls_0/sol1/i>
INFO: Copied 8 vhdl file(s) to /tmp/finn_dev_i22amess/code_gen_ipgen_StreamingMaxPool_hls_0_xpewuftl/project_StreamingMaxPool_hls_0/sol1/impl>
ERROR: [Common 17-685] Unable to load Tcl app xilinx::xsim
Caught error:
ERROR: [Common 17-69] Command failed: ERROR: [Common 17-685] Unable to load Tcl app xilinx::xsim

```
I tried to reset tcl with following commands : 

```
vivado -mode tcl
tclapp:reset_tclstore
```
But the problem persists.

## Optional

### Possible fix

I am not sure but it may be an error related to Vivado/Vitis version 2024.1 

### Additional context
I am also working with singularity but I am not sure if it interfers in this step

### Thank you very much in advance for your effort and your help

fpjentzsch, 2024-10-17:
Hi,
it's most likely [this issue](https://adaptivesupport.amd.com/s/article/63253?language=en_US), which is basically a race condition between multiple Vivado instances running on a slow filesystem.

We fixed this on our end by setting `setenv XILINX_LOCAL_USER_DATA no`.

IsmailAM1999, 2024-10-31:
It's working fine thank you very much !

closed 2024-11-01 by fpjentzsch