Xilinx/finn #1223: Error on run-docker.sh on dev branch : Unable to find image 'xilinx/finn:v0.10.1-84-g45a4e4af-dirty.xrt_202220.2.14.354_22.04-amd64-xrt

kalahel, 2024-11-04:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out. 
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
I am currently trying the dev branch of Finn on WSL, with vivado 2024.1 but when I tried running `run-docker.sh` I got this error :
```shell
4.803 Get:70 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 libgl1-mesa-dev amd64 23.2.1-1ubuntu3.1~22.04.2 [6842 B]
4.836 Err:61 http://security.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-generic amd64 5.15.0.122.122
4.836   404  Not Found [IP: 185.125.190.83 80]
4.841 E: Failed to fetch http://security.ubuntu.com/ubuntu/pool/main/l/linux-meta/linux-headers-generic_5.15.0.122.122_amd64.deb  404  Not Found [IP: 185.125.190.83 80]
4.841 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
4.841 Fetched 98.6 MB in 4s (26.7 MB/s)
------

 1 warning found (use docker --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 139)
Dockerfile.finn:88
--------------------
  87 |
  88 | >>> RUN if [ -z "$SKIP_XRT" ];then \
  89 | >>>     apt install -y /tmp/$XRT_DEB_VERSION.deb && \
  90 | >>>     rm /tmp/$XRT_DEB_VERSION.deb; fi
  91 |
--------------------
ERROR: failed to solve: process "/bin/sh -c if [ -z \"$SKIP_XRT\" ];then     apt install -y /tmp/$XRT_DEB_VERSION.deb &&     rm /tmp/$XRT_DEB_VERSION.deb; fi" did not complete successfully: exit code: 100
Unable to find image 'xilinx/finn:v0.10.1-86-ga054165e.xrt_202220.2.14.354_22.04-amd64-xrt' locally
docker: Error response from daemon: pull access denied for xilinx/finn, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
```
### Steps to Reproduce

1. Launch an Ubuntu WSL
2. Clone the FINN repository
4. Checkout the dev branch, with commit hash: 50045ffa77cf60c566dc3e309dff74598a038802
5. Start windows docker desktop
6. Run `export FINN_XILINX_VERSION=2024.1`
7. Run `export FINN_XILINX_PATH=/mnt/c/Tools/Xilinx_linux`
8. Start the docker container with the command: `./run-docker.sh`

### Additional context
I don't have this problem on the `main` branch.

fpjentzsch, 2024-11-04:
Hi,

can you try again on the latest dev branch after setting the environment variable `FINN_DOCKER_NO_CACHE=1`? Sometimes a fresh rebuild of the container is required. Afterwards you can remove the variable again to save time during future container launches.

kalahel, 2024-11-04:
Hi, thanks for your reply.

I tried it and same result.

This is the full log if it can help :
[log_finn.txt](https://github.com/user-attachments/files/17619403/log_finn.txt)

fpjentzsch, 2024-11-04:
The Docker output still says "CACHED" for most steps. Are you sure you are on the latest dev branch, this env var was added only last week.

kalahel, 2024-11-05:
I think so, because if I run a `git status` this is the result :
```
On branch dev
Your branch is up to date with 'origin/dev'.
```

Mingtiao19, 2025-01-18:
> I think so, because if I run a `git status` this is the result :
> 
> ```
> On branch dev
> Your branch is up to date with 'origin/dev'.
> ```

Hi, I encountered the same problem when doing quicktest. Have you solved that problem already?

Andy4Worc, 2025-06-21:
Did you ensure your WSL Vivado application has the device installed? I was trying to run the pynq-z1/z2 from the examples. This requires to go to the vivado installer (can be accessed from the main Vivado application menu) and ensuring you have "SoC" devices. I am not sure which device you are targeting or what this "image" is, but could be related I'd guess.

closed 2025-06-24 by kalahel