Xilinx/finn #1226: Non-blocking assignments used in Q_srl.v combinational logic blocks

jsmonson, 2024-11-07:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [x] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
The Q_srl.v hdl module using non-blocking assignments in combinational logic blocks. This generates a warning for each occurrence (about 30) during RTL sim in the form: 

Q_srl.v:189:21: Non-blocking assignment '<=' in combinational logic process: ... This will be executed as a blocking assignment '='!
  189 |    shift_en_   <= 1'bx;
         |                     ^~

## Details
This is not necessary is bug as the code appears to work correctly but the overabundance of warnings make it difficult to see other warnings that may be relevant. 

### Steps to Reproduce
Add what needs to be done to reproduce the bug. Add code examples where useful
and make sure to include the resulting ONNX files, and the commit hash you are working on.

1. Clone the FINN repository
2. Checkout the dev branch, with commit hash: a054165eba7c91832387254201027ce5748c3068
3. Start the docker container with the command: ./run-docker.sh 
4. cd tests/fpgadataflow 
5. -k pytest -s -k test_fpgadataflow_fifo_rtlsim

### Expected behavior
The RTL sim runs without the above noted warnings.

### Actual behavior
The RTL sim has the above noted warnings.

## Optional

### Possible fix
Change the non-blocking assignments to blocking assignments in combinational logic. 

### Additional context
Add any other context about the problem here.

auphelia, 2024-11-11:
Hi @jsmonson, thanks for bringing this up and creating the issue! If you like to contribute the fix yourself, please create a PR by following the guidelines in https://github.com/Xilinx/finn/blob/dev/CONTRIBUTING.md 
Otherwise, we'll address this in the coming weeks.

auphelia, 2025-02-20:
Solved in this PR: https://github.com/Xilinx/finn/pull/1251

closed 2025-02-20 by auphelia