Xilinx/finn #1228: Pynq ZU Support for TFC end2end example

BrandonPanuccio, 2024-11-13:
Hey! @auphelia  I saw you were the most recent individual to update the tfc end2end example. I have a ZU board and was interested in expanding the ZynqBuild function to be able to run and eventually create a .bit file to be ran on the board. I have completed this example with my Z2 board, and was wondering if you or anyone had any tips/advice on how to broaden the number of boards supported by FINN. I began editing the basic.py to add the pynq_part_map and pynq_native_port_width using information from the AMD website. Now I am working on the make_zynq_proj.py file. Once again, any and all advice helps!

Thanks

BrandonPanuccio, 2024-11-22:
Nvmd, I figured it out. If anyone needs help adding more boards let me know!

auphelia, 2024-11-26:
Sorry, I didn't have the chance to reply earlier. Glad to hear that you figured it out! 
Would you like to share with the community how you enabled the ZU board? Maybe in a [Show & Tell](https://github.com/Xilinx/finn/discussions/categories/show-and-tell)?

BrandonPanuccio, 2024-11-26:
Thanks for the reply. As mentioned, I have figured out a manual addition process for new boards. I am currently working on an automation process with 3 other Villanova Computer Engineers. I will gladly share both once the documentation/code is written.