Xilinx/finn #1272: Synthesis failed during running cnv_end2end_example

AliMirhosseini, 2025-02-03:
I am working with FINN and successfully ran the TFC example, but the cnv_end2end_example fails during synthesis.
Here is the error message from the Jupyter Notebook: 
[image]

I am using Vitis 2023.2 and WSL2 Ubuntu 20.04.
Below is a screenshot of the Vivado design run:

[image]

Additionally, here are the logs from Vivado:
[runme.log](https://github.com/user-attachments/files/18641874/runme.log)
[vivado.log](https://github.com/user-attachments/files/18641873/vivado.log)

Please let me know if you need any more information.

Reza-Q, 2025-02-09:
Increase your RAM size, this problem is due to Low RAM Space in your system.

AliMirhosseini, 2025-02-09:
> Due to your log report this problem is because of Low RAM Space in your system.
Yes, it works properly after increasing the RAM.  
Thanks.  
For the record, I increased the RAM from 16GB to 32GB, and I think at least 20GB is needed.

closed 2025-02-09 by AliMirhosseini