Xilinx/finn #1276: [VitisLink] Unsupported bitwidth 4096 when linking multiple StreamingDataflowPartitions together

bwintermann, 2025-02-11:
## Quick summary
When splitting the graph into multiple StreamingDataflowPartitions, in the linking stage, occasionally the error "Port s_axis_0 on component StreamingDataflowPartition_XYZ has unsupported data width 4096 for direct connection" appears. The used board is an Alveo U280 with Vitis 2022.2, although the error also appears in later versions.

## Details
I know this issue might  not be of direct concern for most users here, but wanted to ask it anyways. I have custom transformations that split up the dataflow graph into multiple SDPs by using a custom lambda function in CreateDataflowPartition. These are packaged into seperate .xo later on and connected in the Vitis linker config using direct streaming connections (via a modified VitisBuild flow). This works fine most of the time, unless the mentioned error appears. I checked the steps ONNX file and the Verilog wrapper in the vivado_stitched_ip dir, and I get a folded shape with innermost dimension 1024 on UINT4 - thus the 4096 bit width. However I am unsure why this error occurs. As far as I know 4096 is still in spec for Vitis AXI Streams and even if the datawidths between the one SDP output and the other SDP input wouldn't match, Vitis should still be able to insert a DWC automatically. It is also a power of 2, so this can't be the issue either. I searched for this specific error ([CF2XD 83-2220]) but found nothing relevant. AP_INT_MAX_W in the code_gen of the first node in the corresponding SDP is also set to 4096 exactly.

I would be thankful for any hints or ideas on why this error occurs.

fpjentzsch, 2025-02-17:
Might be related to the following limitations of the AXI4-Stream Infrastructure IP Suite? Although 4096 would still be in spec for the FIFO and the DWC would only be an issue if you tried to down-convert the width *to* 4096 or the internal streams would get too wide in case of a "non-integer multiple byte ratio conversion".

AXI4-Stream Data FIFO:

> Total FIFO data width of up to 4096 bits.

AXI4-Stream Data Width Converter:

> Up-conversion features:
> • Range: Input 1-256 Bytes, Output 2-512 Bytes
> Down-conversion features:
> • Range: Input 2-512 bytes, output 256-1 bytes

bwintermann, 2025-02-17:
This could be. I will look into it and try to only split the SDPs where the boundary crossing would not reach the 4096 bit width. Although I would still expect 4096 to work for the DWC.