Xilinx/finn #1278: Problems with example notebooks

OscarWahllof, 2025-02-19:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [ ] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [ ] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
I am unable to run the fin example notebooks. Running the ./run-docker.sh notebook and for example the end2end_example/bnn-pynq/cnv_end2end_example.ipynb silently dies.

## Details
Running the end2end_example/bnn-pynq/cnv_end2end_example.ipynb, the hardware generation fails. Looking at the logs in tmp/finn_dev_<user> I see that the in the logs for code_gen_ipgen_StreamingDataflowPartition_0_IODMA_hls_0_rs0gq2wf fails to export RTL as Vivado IP. The ip generation seems to succeed but the export fails.

Excerpt of the log:
INFO: [RTGEN 206-100] Finished creating RTL model for 'StreamingDataflowPartition_0_IODMA_hls_0'.
INFO: [RTMG 210-285] Implementing FIFO 'dma2dwc_U(StreamingDataflowPartition_0_IODMA_hls_0_fifo_w64_d2_S)' using Shift Registers.
INFO: [RTMG 210-285] Implementing FIFO 'numReps_c_U(StreamingDataflowPartition_0_IODMA_hls_0_fifo_w32_d2_S)' using Shift Registers.
INFO: [RTMG 210-285] Implementing FIFO 'start_for_StreamingDataWidthConverter_Batch_64u_8u_384u_U0_U(StreamingDataflowPartition_0_IODMA_hls_0_start_for_StreamingDataWidthConverter_Batch_64u_8u_384u_U0)' using Shift Registers.
INFO: [HLS 200-111] Finished Creating RTL model: CPU user time: 0.04 seconds. CPU system time: 0.01 seconds. Elapsed time: 0.04 seconds; current allocated memory: 851.766 MB.
INFO: [HLS 200-111] Finished Generating all RTL models: CPU user time: 0.16 seconds. CPU system time: 0 seconds. Elapsed time: 0.17 seconds; current allocated memory: 854.637 MB.
INFO: [HLS 200-1603] Design has inferred MAXI bursts and missed bursts, see Vitis HLS GUI synthesis summary report for detailed information.
INFO: [HLS 200-111] Finished Updating report files: CPU user time: 0.46 seconds. CPU system time: 0.02 seconds. Elapsed time: 0.48 seconds; current allocated memory: 860.660 MB.
INFO: [VHDL 208-304] Generating VHDL RTL for StreamingDataflowPartition_0_IODMA_hls_0.
INFO: [VLOG 209-307] Generating Verilog RTL for StreamingDataflowPartition_0_IODMA_hls_0.
INFO: [HLS 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [HLS 200-789] **** Estimated Fmax: 136.78 MHz
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 7.35 seconds. CPU system time: 0.86 seconds. Elapsed time: 8.21 seconds; current allocated memory: 112.809 MB.
INFO: [HLS 200-1510] Running: export_design -format ip_catalog 
INFO: [IMPL 213-8] Exporting RTL as a Vivado IP.
ERROR: [Common 17-182] exec failed: Permission denied
ERROR: [IMPL 213-28] Failed to generate IP.
INFO: [HLS 200-111] Finished Command export_design CPU user time: -9.58 seconds. CPU system time: -1.23 seconds. Elapsed time: 0.96 seconds; current allocated memory: 6.617 MB.

With the current versions and install I no longer get a notebook error, but earlier I also got a VLNV error crash there indicating that the ip was not found in the stitch stage. Not sure why that no longer happens, now it simply stalls.

I have have reinstalled vivado/vitis several times, as far as I can tell the installation is fine, I cannot find any permission problems for the generated folders of for vivado.
Running Ubuntu 24.04.1 LTS
Have tried both Vivado 2022.2 and Vivado 2024.2
I have followed the getting started instruction on the wiki and have docker configured to run without root, the following environment variables are set (my install is located /home/<user>/Xilinx, have tried just in the home folder also) :

export FINN_XILINX_PATH=/home/<user>/Xilinx
export FINN_XILINX_VERSION=2022.2
export VIVADO_PATH=/home/<user>/Xilinx/Vivado/2022.2/settings64.sh

The quicktest finishes with:
977 passed, 16 skipped, 4 xfailed, 2 xpassed, 76259 warnings in 107.08s

Commit is 9d299689, just cloned the main of the repo from the getting started guide.

I am intending to work with FINN for accelerator design for a Kria KR260, as far as I know there are no pre-built examples for this platform, but the examples should still run to the last deployment step still?

I feel like I have tried everything short of trying it on a different machine. Is this a FINN issue or a Vivado/Vitis issue? Any insight you may have would be useful as I am out of ideas. I have searched the web and tried everything I could find unfortunately.

### Steps to Reproduce
See above.

### Expected behavior
N/A

### Actual behavior
N/A

auphelia, 2025-02-20:
Hi @OscarWahllof,
That looks like a Vivado issue. Are you working with the dev branch? Did you already change the notebook to use the Kria board as development target? Or is this failure also with the default setting (Pynq-Z1)?

> With the current versions and install I no longer get a notebook error, but earlier I also got a VLNV error crash there indicating that the ip was not found in the stitch stage. Not sure why that no longer happens, now it simply stalls.

I am not sure how to read this part, do you mean current versions of finn or Vivado?

closed 2025-05-30 by OscarWahllof

auphelia, 2025-05-30:
Thank you for updating the issue and explaining the solution @OscarWahllof !