Xilinx/finn #1286: C driver for the IP

m7md5303, 2025-02-25:
I think it would be very useful if you could add the feature of generating a C driver for the output IP. I have a problem with integrating it with a pre-built C project on Vitis. Is there a way to work around this issue or it is meant to make the inference specific for PYNQ?
Thanks in advance

fpjentzsch, 2025-02-26:
Hi,

you could look how the FINN Python driver interacts with the IDMA/ODMA AXI-Lite peripherals [here ](https://github.com/Xilinx/finn/blob/cbf3636b3ee0f08fa3908aee6578229d4193d4c9/src/finn/qnn-data/templates/driver/driver_base.py#L337) and replicate that behavior in C.

There is also this PR that adds a C++ driver for Alveo platforms: https://github.com/Xilinx/finn/pull/989

m7md5303, 2025-02-26:
@fpjentzsch Thank you. I hope it works