Xilinx/finn #1295: Trouble with running run-docker.sh: permission denied and not finding modules

Hermanstray, 2025-03-07:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x ] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out. (commit 7ccbb15e)
- [x ] Check that the issue hasn't already been reported, by checking the currently open issues.

### Brief summary 
I try running the run-docker.sh quickstart, and end up failing 91 tests with errors because of not finding dependencies such as qonnx, Brevitas, finn and pyverilator. I get a permission error with moving the pyproject.toml file associated with the qonnx dependency despite having granted permissions of the whole finn folder to my user. 

## Details
This I believe is the root of the problem: For some reason I cannot move this pyproject.toml file associated with the qonnx dependency. I have made sure that my users has +rwx permissions for the whole folder so I don't understand where this stems from. I am new to both Linux and Docker so this might be an easy fix.

[image]

I also get a bunch of these errors complaining about deprecated setuptools and other packages. I have tried upgrading them to newer version that does not conflict with dependants like Brevitas.

[image]

Finally, I get all these errors from the pystests that occur because of dependency issues. The packages are all in my file system though.

[image]

I have tried to look through all the earlier issues but I cannot seem to find the same bug.

### Steps to Reproduce
I am using Ubuntu 2022.04, Vivado/Vitis 2022.2, Docker in rootless mode. I have done just as explained in the quickstart guide. I have put the required enviroment variables in the bashrc script.

Any help would be much appeciated as I have exhausted my options. I have tried clean installing everything several times to no result. 
Thanks

Mounice, 2025-03-13:
I had the same problem and the only work around that I was able to find, was to set the flag : ${FINN_DOCKER_RUN_AS_ROOT="1"} in the run-docker.sh file. In that way the permissions are granted as root and the tests can be completed succesfully. However I don't think this is a desirable solution. Hope the developers can give you a more specific way to solve this.

Hermanstray, 2025-03-13:
Thanks a lot @Mounice !

Now the tests in quicktest pass, although with a lot of generated warnings about DataTypes rounding and tensor representation.

auphelia, 2025-03-26:
Thanks @Mounice for your help and @Hermanstray thanks for your question!
The warnings can be seen more as notifications for the user, rather than critical warnings. Because we are doing the compute in software for verification in these tests, there are sometimes small rounding errors. So, for example the output of a node is known to be int4 but the output from verification is a float of 1.00000004. If the variation is small enough, we would round this to 1 to avoid error propagation through layers. A warning is raised to notify the user, that there has been rounding applied.

closed 2025-04-11 by auphelia