Xilinx/finn #1306: Folding configuration - resType not taken in account

kalahel, 2025-03-26:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [X] Test that the bug appears on the current version of the master-branch. 
- [X] Check that the issue hasn't already been reported, by checking the currently open issues.
- [X] If there are steps to reproduce the problem, make sure to write them down below.

## Quick summary
I'm trying to change the folding configuration of the MobileNetV1 but changing `resType` from `dsp` to `lut` does not affect the count of `dsp` for those layers in the `estimate_layer_resources.json`.

## Details
I slightly modified `finn-examples/build/mobilenet-v1/build.py` to only generate output for the `ZCU102`. I modified the `ZCU102_folding_config.json`, changed the `resType` of the first 6 MVAU from `dsp` to `lut`. as such :
```json
  "MVAU_rtl_5": {
    "PE": 32,
    "SIMD": 8,
    "ram_style": "auto",
    "resType": "lut",
    "mem_mode": "internal_decoupled",
    "runtime_writeable_weights": 0
  },
```
I only run those steps : 
```python
  step_mobilenet_streamline,
  step_mobilenet_lower_convs,
  step_mobilenet_convert_to_hw_layers_separate_th,
  "step_create_dataflow_partition",
  "step_specialize_layers",
  "step_apply_folding_config",
  "step_minimize_bit_width",
  "step_generate_estimate_reports",
```
After all those steps are completed I check the `estimate_layer_resources.json` but it gives the same values for `dsp` as when the resType was `dsp` :
``` 
  "MVAU_rtl_5": {
    "BRAM_18K": 29,
    "BRAM_efficiency": 0.4904214559386973,
    "LUT": 0,
    "URAM": 0,
    "URAM_efficiency": 1,
    "DSP": 64
  },
```

### Steps to Reproduce

1. Clone the FINN repository
2. Checkout the main branch, with commit hash: [9d29968](https://github.com/Xilinx/finn/commit/9d299689f2ec0895f208b8bfe3bcdcf6f450181a)
3. Checkout [finn-examples](https://github.com/Xilinx/finn-examples)
4. Modify  `finn-examples/build/mobilenet-v1/build.py` to only run steps described in **Details** section
5. Modify the `estimate_layer_resources.json` from finn-example, change `MVAU_rtl_0` to `MVAU_rtl_5`, `resType` from `dsp` to `lut`
6. Start the docker container with the command (from the finn repo): `./run-docker.sh build_custom [path]/finn-examples/build/mobilenet-v1`
7. Check report for changes

Somebody knows why this change is not taken in account ?
Also, is there a simpler way to run this script without having to restart docker image each time and calling `build_custom` ?

auphelia, 2025-03-26:
Hi @kalahel,

We only have optimized RTL compute with DSPs for MVAU RTL, the resType is available as a node attribute because we are planning on having an RTL MVAU using LUTs as well in the future. For now, this node attribute only influences the HLS MVAU.

kalahel, 2025-03-26:
Hi @auphelia

Thank you for your answer it's clearer now.