Xilinx/finn #1309: Folding configuration - MVAU_rtl ram_style distributed not converting properly

kalahel, 2025-03-31:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [X] Test that the bug appears on the current version of the master-branch.
- [X] Check that the issue hasn't already been reported, by checking the currently open issues.
- [X] If there are steps to reproduce the problem, make sure to write them down below.

## Quick summary
I'm trying to change the folding configuration of the MobileNetV1 but changing `ram_style` from `auto` to `distributed` result in a `lut` and `bram` utilization of 0 in the `estimate_layer_resources.json`.

## Details
I slightly modified `finn-examples/build/mobilenet-v1/build.py` to only generate output for the `ZCU102`. I modified the `ZCU102_folding_config.json`, changed the `ram_style` of the  MVAU from `auto` to `distributed`. as such :
```json
  MVAU_rtl_14": {
    "PE": 4,
    "SIMD": 16,
    "ram_style": "distributed",
    "resType": "dsp",
    "mem_mode": "internal_decoupled",
    "runtime_writeable_weights": 0
  },
```
I only run those steps : 
```python
  step_mobilenet_streamline,
  step_mobilenet_lower_convs,
  step_mobilenet_convert_to_hw_layers_separate_th,
  "step_create_dataflow_partition",
  "step_specialize_layers",
  "step_apply_folding_config",
  "step_minimize_bit_width",
  "step_generate_estimate_reports",
```

After all those steps are completed I check the `estimate_layer_resources.json` but the resources estimation for `lut` and `bram` show 0 utilization :
``` json
# Now :
  "MVAU_rtl_14": {
    "BRAM_18K": 0,
    "BRAM_efficiency": 1,
    "LUT": 0,
    "URAM": 0,
    "URAM_efficiency": 1,
    "DSP": 16
  },

# Previously was :
"MVAU_rtl_14": {  
  "BRAM_18K": 240,  
  "BRAM_efficiency": 0.9259259259259259,  
  "LUT": 0,  
  "URAM": 0,  
  "URAM_efficiency": 1,  
  "DSP": 16  
},
```

### Steps to Reproduce
1. Clone the FINN repository
2. Checkout the main branch, with commit hash: [9d29968](https://github.com/Xilinx/finn/commit/9d299689f2ec0895f208b8bfe3bcdcf6f450181a)
3. Checkout [finn-examples](https://github.com/Xilinx/finn-examples)
4. Modify  `finn-examples/build/mobilenet-v1/build.py` to only run steps described in **Details** section
5. Modify the `estimate_layer_resources.json` from finn-example, change `MVAU_rtl_14`, `ram_style` from `auto` to `distributed`
6. Start the docker container with the command (from the finn repo): `./run-docker.sh build_custom [path]/finn-examples/build/mobilenet-v1`
7. Check report for changes

Somebody knows why the report shows 0 utilization for both resources ?

fpjentzsch, 2025-03-31:
Hi,

BRAM utilization of 0 is expected if you switch to `distributed`.
LUT utilization is always reported as 0 for the RTL MVAU (see [here](https://github.com/Xilinx/finn/blob/0a29f812ab544e21803325a9daf42630215dfb93/src/finn/custom_op/fpgadataflow/rtl/matrixvectoractivation_rtl.py#L131)). Resource estimation in FINN is only a best-effort and very inaccurate in many places. Seems like no one bothered to implement an estimation formula for the RTL MVAU yet, probably because it always uses DSP compute and thus has a generally smaller LUT footprint.

kalahel, 2025-03-31:
Hi,

Thanks a lot for the answer and the link.
I'll stick to using DSP then.

Best regards,
Mathieu