Xilinx/finn #1320: step syntheisze bitfile doesn't finish

Thomastw123, 2025-04-12:
When i run the step synthesize bitfile for my custom pointpillars model, the step doesn't finish or generate any outputs in the bitfile folder. This step has been running for 7 hours now and still hasn't finished or generated any outputs. Has anyone encountered the same issue, and if so how can i solve it?

fpjentzsch, 2025-04-23:
Hi, you'll need to look into the underlying Vivado project logs (in FINN_HOST_BUILD_DIR) to see where it is hanging.

auphelia, 2025-05-09:
Hi @Thomastw123 and thanks @fpjentzsch ,

Did you have a chance to try @fpjentzsch 's suggestion? Depending on the network and your compute resources, it is not uncommon that the last step (synthesize the bitfile) may take hours.

Thomastw123, 2025-05-09:
Yea, I fixed the problem. The insert IODMA script wasn’t compatible with my multi-output network. So I had to do some changes to the script for it to work on my network. Previously it went into an infinite loop trying to insert IODMA between branches but failed and restarted every time.

auphelia, 2025-05-09:
That sounds interesting, did you work with the pynq driver and did adjustments there as well?
And is that something you consider creating a PR for to dev? 
If yes, here is the link to the contribution guidelines: https://github.com/Xilinx/finn/blob/main/CONTRIBUTING.md

Thomastw123, 2025-05-09:
Finn generated the driver successfully and the deployment package, but I haven’t had the chance to modify the driver or test the deployment package. Once I deploy it on the board and confirm everything goes smoothly, I will update you and consider creating a PR. I also added a custom create_dataflow_partition step for it to work on my network (PointPillars for LIDAR object detection).