Xilinx/finn #1330: Finn can't get get the license (No such file or directory: '/tmp/finn_dev_thesis/synth_out_of_context_unxu_jk8/results_finn_design_wrapper/res.txt')

abdosalem490, 2025-05-08:
## Prerequisites
Dev Commit: **cbf3636b3ee0f08fa3908aee6578229d4193d4c9**.
Operating System: Ubuntu 24.04 LTS
Board: ZCU102

## Quick summary
Finn can't see the license file even though the 
`export FINN_DOCKER_EXTRA=" -v /path/to/licenses:/path/to/licenses -e XILINXD_LICENSE_FILE=/path/to/licenses "`
is executed. as specified in the [Vivado/Vitis license section](https://finn.readthedocs.io/en/latest/getting_started.html#vivado-vitis-license). I keep getting the following error in my python script 
```
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/finn_dev_thesis/synth_out_of_context_unxu_jk8/results_finn_design_wrapper/res.txt'
```

I opened Vivado log file and I see the following error:
```
A valid license was not found for feature 'Synthesis' and/or device 'xczu9eg'. Please run the Vivado License Manager for assistance in determining
which features and devices are licensed for your system.
```

## Details

I have the license stored at the following directory: `/eda/Vivado/2024.2/data/ip/core_licenses` with the names:
- vivado.log
- runme.log

Here is the last couple of line of the Vivado.log:
```
# source headers.tcl
## set file "$origin_dir/dummy_nachiket_fooling_zsh_for_loops.h"
## set file [file normalize $file]
## set file_obj [get_files -of_objects [get_filesets sources_1] [list "*$file"]]
## set_property "file_type" "Verilog Header" $file_obj
# set obj [get_filesets sources_1]
# set_property "top" "$argv" $obj
# if {[string equal [get_filesets -quiet constrs_1] ""]} {
#   create_fileset -constrset constrs_1
# }
# set obj [get_filesets constrs_1]
# set file "[file normalize "$origin_dir/$argv.xdc"]"
# set file_added [add_files -norecurse -fileset $obj $file]
# set file "$origin_dir/$argv.xdc"
# set file [file normalize $file]
# set file_obj [get_files -of_objects [get_filesets constrs_1] [list "*$file"]]
# set_property "file_type" "XDC" $file_obj
# set obj [get_filesets constrs_1]
# if {[string equal [get_filesets -quiet sim_1] ""]} {
#   create_fileset -simset sim_1
# }
# set obj [get_filesets sim_1]
# set obj [get_filesets sim_1]
# set_property "top" "$argv" $obj
# set_property "xelab.nosort" "1" $obj
# set_property "xelab.unifast" "" $obj
# set_property strategy "Vivado Synthesis Defaults" [get_runs synth_1]
# set_property strategy "Flow_PerfOptimized_high" [get_runs synth_1]
# set_param synth.elaboration.rodinMoreOptions {rt::set_parameter ignoreVhdlAssertStmts false}
# set obj [get_runs synth_1]
# set_property -name {steps.synth_design.args.more options} -value {-mode out_of_context} -objects $obj
# set_property STEPS.SYNTH_DESIGN.ARGS.RETIMING true [get_runs synth_1]
WARNING: [Common 17-599] Property 'STEPS.SYNTH_DESIGN.ARGS.RETIMING' is deprecated for object type 'run'. Please use '-global_retiming' instead.
# current_run -synthesis [get_runs synth_1]
# set_property strategy "Vivado Implementation Defaults" [get_runs impl_1]
# current_run -implementation [get_runs impl_1]
# puts "INFO: Project created:vivadocompile"
INFO: Project created:vivadocompile
# launch_runs -jobs 8 impl_1 -to_step route_design
[Thu May  8 03:48:56 2025] Launched synth_1...
Run output will be captured here: /tmp/finn_dev_thesis/synth_out_of_context_unxu_jk8/results_finn_design_wrapper/vivadocompile/vivadocompile.runs/synth_1/runme.log
[Thu May  8 03:48:56 2025] Launched impl_1...
Run output will be captured here: /tmp/finn_dev_thesis/synth_out_of_context_unxu_jk8/results_finn_design_wrapper/vivadocompile/vivadocompile.runs/impl_1/runme.log
# wait_on_run impl_1
[Thu May  8 03:48:56 2025] Waiting for impl_1 to finish...
[Thu May  8 03:49:16 2025] impl_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'synth_1'
 'synth_1' run failed with below errors.
ERROR: [Common 17-345] A valid license was not found for feature 'Synthesis' and/or device 'xczu9eg'. Please run the Vivado License Manager for assistance in determining
which features and devices are licensed for your system.
Resolution: Check the status of your licenses in the Vivado License Manager. For debug help search Xilinx Support for "Licensing FAQ". If you are using a license server, verify that the license server is up and running a version of the xilinx daemon that is compatible with the version of Xilinx software that you are using. Note: Vivado 2021.1 and later versions require upgrading your license server tools to the Flex 11.17.2.0 versions. Please confirm with your license admin that the correct version of the license server tools are installed.

wait_on_runs: Time (s): cpu = 00:00:09 ; elapsed = 00:00:20 . Memory (MB): peak = 1460.258 ; gain = 0.000 ; free physical = 3432 ; free virtual = 29476
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run impl_1"
    (file "finn_design_wrapper.tcl" line 156)
INFO: [Common 17-206] Exiting Vivado at Thu May  8 03:49:16 2025...
```
I am afraid that I can't share the source code due to NDA I have signed. 

Any ideas how to debug such a problem?

auphelia, 2025-05-08:
Hi @abdosalem490 ,

Can you check from inside the container if the path was correctly mounted? So, just executing `./run-docker.sh` after the env vars are set and then navigating to the folder you have specified? 
The env var `FINN_DOCKER_EXTRA` just extends the docker command in the run-docker.sh shell script here: https://github.com/Xilinx/finn/blob/dev/run-docker.sh#L271 and `-v` indicates a volume mount and `-e` an env setting inside the container. You could verify that both are propagated correctly.

abdosalem490, 2025-05-08:
@auphelia 

I have tried to modify the "run-docker.sh" file by adding the following line:
`FINN_DOCKER_EXTRA+="-v /eda/Vivado/2024.2/data/ip/core_licenses/Xilinx.lic:/eda/Vivado/2024.2/data/ip/core_licenses/Xilinx.lic -e XILINXD_LICENSE_FILE=/eda/Vivado/2024.2/data/ip/core_licenses/Xilinx.lic"`

However, when I ran my code I always get the following error:
[image]

I guess it has something to do with the license as indicated in the aforementioned issue before.
I opened the same docker image using the command `docker exec -it <image_id> bash` and in the new terminal I typed the command 
`env | grep FINN`
and this is the output:
```
FINN_ROOT=/home/thesis/finn
FINN_BUILD_DIR=/tmp/finn_dev_thesis
```

Even though I have edited the "run-docker.sh" to define the missing environmental variables as follows:
```
FINN_XILINX_PATH='/eda'
FINN_XILINX_VERSION=2024.2
PLATFORM_REPO_PATHS='/eda'
```

It didn't change anything. I just can't get my head around why this happens and how to debug it?

auphelia, 2025-05-09:
Hi @abdosalem490 ,

It shouldn't be necessary to change the run-docker.sh. 
Does Vivado with the licenses work outside the FINN docker container?

Instead of changing the run-docker.sh, I would suggest the following (the spaces in the env vars are important):
```
export FINN_XILINX_PATH="/eda"
export FINN_XILINX_VERSION="2024.2"
export PLATFORM_REPO_PATHS="/eda"
export FINN_DOCKER_EXTRA=" -v /eda/Vivado/2024.2/data/ip/core_licenses:/eda/Vivado/2024.2/data/ip/core_licenses -e XILINXD_LICENSE_FILE=/eda/Vivado/2024.2/data/ip/core_licenses "
git clone https://github.com/Xilinx/finn.git
cd finn
git checkout dev
bash run-docker.sh
```
Then wait until the docker container is open and if Vivado was correctly found. There should be a green print out if everything was found correctly. Then you can check if the path was correctly mounted and the env var was correctly set:
```
ls /eda/Vivado/2024.2/data/ip/core_licenses/
echo $XILINXD_LICENSE_FILE
```

If the last two show the license folder and the env var prints the expected license path, then everything got correctly propagated inside the Docker container and it is not an issue with the container setup but with the license itself.

Note: Please be aware that we're considering Vivado 2022.2 to be stable with FINN, while 2024.2 should work but might show some issues, I would recommend using `dev` with 2024.2 .

abdosalem490, 2025-05-11:
hi @auphelia,
Vivado works alone normally, problems only occurs when using Finn. I have solved the problem by going to "help -> Manage License -> Manage License Search Path" only to discover that "XILINXD_LICENSE_FILE" variable have the format of "<num>@<ip_address>" and that it's a server based license not a local one.

Anyhow, thanks a lot for the help, it was just quite a naive mistake from my end. Sorry for any disturbance I may have caused.

closed 2025-05-11 by abdosalem490