Xilinx/finn #1351: FINN and ONNX output do not match in 2-import-into-finn-and-verify.ipynb example notebook

SanderDebets, 2025-06-02:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [ Y] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [ Y] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ Y] If there are steps to reproduce the problem, make sure to write them down below.
- [ N] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
I am running the "2-import-into-finn-and-verify.ipynb" notebook located at notebooks/end2end_example/cybersecurity. In the final step, where the ONNX output is compared to the FINN output. Only 52 out of 100 outputs are equal here, while I assume this is supposed to be 100 out of 100. 

## Details
I have made one small modification to the notebook: 

In the first cell in section 4, i have changed the following line:
import finn.core.onnx_exec as oxe
to:
from qonnx.core.onnx_exec import execute_onnx

and 
output_dict = oxe.execute_onnx(model_for_sim, input_dict)
to:
output_dict = execute_onnx(model_for_sim, input_dict)

This is to fix an error I was getting related to TKinter, which prevented me from running the notebook. 

Unfortunately, github is not allowing me to upload any ONNX files to this issue

auphelia, 2025-06-04:
Hi @SanderDebets ,

I checked the notebooks on the dev branch and couldn't see the mismatch that you described. Are you using main or dev? And are you running the notebooks with our docker setup or do you have another setup?

At this point in the flow, it shouldn't matter if you're using qonnx or finn execution, but can you explain why you needed to switch?

SanderDebets, 2025-06-05:
hi @auphelia 

My bad, I was on the main branch. Switching to the dev branch fixed this issue!
Additionally, i was running on my own setup instead of the docker setup, because i didn't realize at the time of writing that this was the preferred way of running the notebooks.
When trying to run bash ./run-docker.sh notebook to run the docker setup i got an error saying XRT installation was not found. 
Trying to install this has been quite troublesome. I'm currently trying to do this on Ubuntu 24.04, and heard I might have an easier time on Ubuntu 22.04. Do you perhaps know if this is the case? Or any other tips for this issue? Thanks!
(I could share error messages etc, but I realize this is already a bit off the original topic)

SanderDebets, 2025-06-05:
Switching to Ubuntu 22.04 (Running WSL2 so made a fresh install of that with 22.04) seems to have fixed all my issues! 
Able to start a docker container with "bash ./run-docker.sh notebook" and run all the notebooks without any error. I'll close this issue for now.

closed 2025-06-05 by SanderDebets