Xilinx/finn #1368: Exception: Synthesis failed, no bitfile found.

Andy4Worc, 2025-06-17:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x ] Test that the bug appears on the current version of the dev-branch. Make sure to include the commit hash of the commit you checked out.
- [ x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ x] If there are steps to reproduce the problem, make sure to write them down below.
- [ x] If relevant, please include the ONNX files, which were created directly before and/or after the bug.

## Quick summary
The notebook example given, "end2end_example/bnn-pynq/cnv_end2end_example" fails synthesis. Either the bitstream is not created or the project 'impl_1' has not been started, depending on whether your looking at the python error or the logfile error respectively.

## Details
Please add to the following sections to describe the bug as accurately as possible.

### log file:

INFO: [IP_Flow 19-6924] IPCACHE: Running cache check for IP inst: top_smartconnect_0_0
INFO: [IP_Flow 19-6924] IPCACHE: Running cache check for IP inst: top_xbar_0
[Tue Jun 17 04:27:53 2025] Launched top_auto_pc_0_synth_1, top_StreamingDataflowPartition_1_0_synth_1, top_idma0_0_synth_1, top_odma0_0_synth_1, top_rst_zynq_ps_100M_0_synth_1, top_xbar_0_synth_1, top_zynq_ps_0_synth_1, top_smartconnect_0_0_synth_1, synth_1...
Run output will be captured here:
top_auto_pc_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_auto_pc_0_synth_1/runme.log
top_StreamingDataflowPartition_1_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_StreamingDataflowPartition_1_0_synth_1/runme.log
top_idma0_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_idma0_0_synth_1/runme.log
top_odma0_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_odma0_0_synth_1/runme.log
top_rst_zynq_ps_100M_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_rst_zynq_ps_100M_0_synth_1/runme.log
top_xbar_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_xbar_0_synth_1/runme.log
top_zynq_ps_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_zynq_ps_0_synth_1/runme.log
top_smartconnect_0_0_synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/top_smartconnect_0_0_synth_1/runme.log
synth_1: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/synth_1/runme.log
[Tue Jun 17 04:27:53 2025] Launched impl_1...
Run output will be captured here: /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y/finn_zynq_link.runs/impl_1/runme.log
launch_runs: Time (s): cpu = 00:00:14 ; elapsed = 00:00:16 . Memory (MB): peak = 1993.625 ; gain = 230.867 ; free physical = 1321 ; free virtual = 3212
#wait_on_run [get_runs impl_1]
[Tue Jun 17 04:27:53 2025] Waiting for impl_1 to finish...
[Tue Jun 17 04:32:36 2025] impl_1 finished
WARNING: [Vivado 12-13638] Failed runs(s) : 'top_StreamingDataflowPartition_1_0_synth_1'
 'top_StreamingDataflowPartition_1_0_synth_1' run failed with below errors.
wait_on_runs: Time (s): cpu = 00:00:00.64 ; elapsed = 00:04:43 . Memory (MB): peak = 1993.625 ; gain = 0.000 ; free physical = 2057 ; free virtual = 2948
#open_run impl_1
ERROR: [Common 17-69] Command failed: Run 'impl_1' has not been launched. Unable to open
INFO: [Common 17-206] Exiting Vivado at Tue Jun 17 04:32:36 2025...

### python output (warnings and error):
/home/andy/Documents/FINN_dev/finn/src/finn/transformation/fpgadataflow/floorplan.py:107: UserWarning: 32 nodes have no entry in the provided floorplan, SLR was set to -1
  warnings.warn(
/home/andy/Documents/FINN_dev/finn/src/finn/transformation/fpgadataflow/insert_fifo.py:239: UserWarning: Input FIFO for IODMA_hls_0_out0 has depth 2 and won't
                        be created. This may cause RTL simulation issues.
                        
  warnings.warn(
/home/andy/Documents/FINN_dev/finn/src/finn/transformation/fpgadataflow/insert_fifo.py:303: UserWarning: Output FIFO for LabelSelect_hls_0_out0 has depth 2 and won't
                        be created. This may cause RTL simulation issues.
                        
  warnings.warn(
/home/andy/Documents/FINN_dev/finn/src/finn/transformation/fpgadataflow/create_stitched_ip.py:317: UserWarning: First node is not StreamingFIFO or IODMA.
                You may experience incorrect stitched-IP rtlsim or hardware
                behavior. It is strongly recommended to insert FIFOs prior to
                calling CreateStitchedIP.
  warnings.warn(
ERROR: [Common 17-69] Command failed: Run 'impl_1' has not been launched. Unable to open

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[14], line 3
      1 from finn.transformation.fpgadataflow.make_zynq_proj import ZynqBuild
      2 model = ModelWrapper(build_dir+"/end2end_cnv_w1a1_folded.onnx")
----> 3 model = model.transform(ZynqBuild(platform = pynq_board, period_ns = target_clk_ns))

File /home/andy/Documents/FINN_dev/finn/deps/qonnx/src/qonnx/core/modelwrapper.py:145, in ModelWrapper.transform(self, transformation, make_deepcopy, cleanup)
    143 model_was_changed = True
    144 while model_was_changed:
--> 145     (transformed_model, model_was_changed) = transformation.apply(transformed_model)
    146 if cleanup:
    147     transformed_model.cleanup()

File /home/andy/Documents/FINN_dev/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py:347, in ZynqBuild.apply(self, model)
    345     kernel_model.save(dataflow_model_filename)
    346 # Assemble design from IPs
--> 347 model = model.transform(
    348     MakeZYNQProject(self.platform, self.period_ns, enable_debug=self.enable_debug)
    349 )
    351 # set platform attribute for correct remote execution
    352 model.set_metadata_prop("platform", "zynq-iodma")

File /home/andy/Documents/FINN_dev/finn/deps/qonnx/src/qonnx/core/modelwrapper.py:145, in ModelWrapper.transform(self, transformation, make_deepcopy, cleanup)
    143 model_was_changed = True
    144 while model_was_changed:
--> 145     (transformed_model, model_was_changed) = transformation.apply(transformed_model)
    146 if cleanup:
    147     transformed_model.cleanup()

File /home/andy/Documents/FINN_dev/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py:264, in MakeZYNQProject.apply(self, model)
    262 bitfile_name = vivado_pynq_proj_dir + "/finn_zynq_link.runs/impl_1/top_wrapper.bit"
    263 if not os.path.isfile(bitfile_name):
--> 264     raise Exception(
    265         "Synthesis failed, no bitfile found. Check logs under %s" % vivado_pynq_proj_dir
    266     )
    267 deploy_bitfile_name = vivado_pynq_proj_dir + "/resizer.bit"
    268 copy(bitfile_name, deploy_bitfile_name)

Exception: Synthesis failed, no bitfile found. Check logs under /tmp/finn_dev_andy/vivado_zynq_proj_0vxxvd9y

### Steps to Reproduce
Add what needs to be done to reproduce the bug. Add code examples where useful
and make sure to include the resulting ONNX files, and the commit hash you are working on.

1. Clone the FINN repository
2. Checkout the dev branch, with commit hash (June 13th): 6f5aade46c4964ec6bc50ab123584d5e1a01f17f
3. Start the docker container with the command: ./run-docker.sh notebook
4. run the dataflow builder with the following settings: run the "cnv_end2end_example" example as is.

### Expected behavior
I'd expect no failure on the examples given without any modifications.
The fully connected version - "tfc_end2end_example" notebook works completely. I can view in Vivado and have a .zip file with the bistream. I do not have the pynq board yet (I just ordered and can set that up and test in a few days hopefully).
It is just this "CNN" based network that is failing on both the main and dev branches.
The warnings in the compiler output given above, show up with the same type of warnings when running the "tfc" example.

### Actual behavior
The same failure happens on both the dev and main branches of finn.

## Optional

### Possible fix

### Additional context

I made sure to stop all kernels before running each example to ensure I am not trying to have two things happening at once.
The netron outputs look good to me. Feel free to ask for specific files if you think I should include something else. I am still fairly new to FINN, so any help is appreciated. Thanks.

Note: I removed a few spaces in the logfile near "#"s so that it wouldn't look weird in the preview. Otherwise, the outputs are exactly as is.

emirworks, 2025-06-18:
I am facing similar issue as well when running these particular cells in the cybersecurity 3rd notebook:

`
import finn.builder.build_dataflow as build
import finn.builder.build_dataflow_config as build_cfg
import os
import shutil

model_file = model_dir + "/cybsec-mlp-ready.onnx"

final_output_dir = "output_final"

#Delete previous run results if exist
if os.path.exists(final_output_dir):
    shutil.rmtree(final_output_dir)
    print("Previous run results deleted!")

cfg = build.DataflowBuildConfig(
    output_dir          = final_output_dir,
    mvau_wwidth_max     = 80,
    target_fps          = 1000000,
    synth_clk_period_ns = 10.0,
    board               = "Pynq-Z2", # I am using a Pynq-Z2 board
    shell_flow_type     = build_cfg.ShellFlowType.VIVADO_ZYNQ,
    generate_outputs=[
        build_cfg.DataflowOutputType.BITFILE,
        build_cfg.DataflowOutputType.PYNQ_DRIVER,
        build_cfg.DataflowOutputType.DEPLOYMENT_PACKAGE,
    ]
)
`
and 

`
%%time
build.build_dataflow_cfg(model_file, cfg)
`

Andy4Worc, 2025-06-20:
Gotcha. And I'll point my OP here was for he Pynq-Z1 as in the original. Good to know I am not the only one having these issues.

I find it interesting that if I take the working example (tfc...), it works for both the Z1 and Z2. However, if I switch it do the ZCU104, it fails with the same error. I am thinking now that synthesis is failing in the ZynqBuild tool.

If I open up the ZCU104 Vivado project I see synthesis fails with a critical warning about a project file not being made and/or being read-only. If I copy the full project over to a non "/tmp" directory so that it is not read-only, I can run synthesis and it passes without that critical warning. I can go on and build the bitstream too.

I tried making the directory permissions more open, but every time you run the ZynqBuild tool, it builds a new random directory, so I am not sure if that would help or not.

Since I can synthesize in Vivado, that seems good, but I do not have the Pynq Driver and other resizing portions and such that happen later in the Jupyter Notebook example flow, so I am not sure what to do still.

Andy4Worc, 2025-07-12:
Yes, I copied the project outside of my /tmp directory, got write permissions, then launched the vivado project that had been made - this showed an error had occurred when the python code had tried to synthesize it. But in the Vivado project view, I was able to re-synthesize and generate a bit stream error free.

However, only the ZCU version has this error for me. I get the "Z1" and "Z2" to get past this error. The driver doesn't work in this example (separate git issue) regardless of board type. However, I found that I can copy the .hwh and .bit files to my pynq board and use the driver provided in the notebook example running on the pynq board provided by the "finn examples" repo - that driver seems better and simpler.