Xilinx/finn #1402: Unexpected output for the new version

m7md5303, 2025-07-29:
In feb 2025 I have cloned the FINN repo, generated IP for a custom NN, simulated it using system verilog and the output was as expected and the utilization was 16.7%. However, a few days ago when I cloned the repo on a new device and thus having these 5 months updates, the simulation outptut was just zeros and I have found that almost all the internal axistream data signals are zeros. Moreover, the utilization is decreased to 15.9% using the same codes which is unexplainable for me.
Thanks in advance

auphelia, 2025-07-31:
Are you using `dev`?

The reduction in resource usage can be attributed to recent updates in the repository. For instance, if you’re using the HLS MVAU, the default resource mapping now targets LUTs, whereas previously Vivado (and not FINN) would select the default resources. There are also other changes that might have contributed to the lower resource utilization.

Regarding the simulation outputting only zeros, could you clarify which simulation you’re referring to? If you’re running a node-by-node RTL simulation, you can configure FINN to return the full execution dictionary, which allows you to check whether the intermediate tensors are also all zeros. However, when running stitched IP RTL simulation with the full execution dictionary enabled, you’ll see zeros because the simulation no longer has access to the internal tensors. In this case, you can have FINN generate a waveform to track the intermediate tensors.

Could you provide more details on how you’re running the simulation? Also, do you observe only zeros when running the accelerator on the actual board as well?

auphelia, 2025-07-31:
> [@auphelia](https://github.com/auphelia) Isn't it the other way around? Prior to introduction of the RTL/DSP MVAU, the HLS MVAU was set to use LUT for compute by default. Now it is auto, which often causes Vivado to implement compute with DSPs, even though it would be more efficient to use the RTL MVAU if you really wanted to use DSPs. To get the old behavior, one has to set every HLS MVAU resType to "lut" manually.
> 
> [finn/src/finn/custom_op/fpgadataflow/matrixvectoractivation.py](https://github.com/Xilinx/finn/blob/6d2dc98f4831835c7ba15f61e093488f6986a236/src/finn/custom_op/fpgadataflow/matrixvectoractivation.py#L66)
> 
> Line 66 in [6d2dc98](/Xilinx/finn/commit/6d2dc98f4831835c7ba15f61e093488f6986a236)
> 
>  "resType": ("s", False, "auto", {"auto", "lut", "dsp"}),

This is now changed, see this PR: https://github.com/Xilinx/finn/pull/1396

m7md5303, 2025-07-31:
[image]
Github doesn't allow onnx files uploading, here it is on google drive https://drive.google.com/file/d/12NQ2clYYnn5tTfBoQ7p5eg0DyDPHYZvp/view?usp=sharing

m7md5303, 2025-08-03:
@auphelia @fpjentzsch 
New updates:
* I have checked the .dat file and the weights.npy file for the MVAU_0 block which seemed to be the start point of the zero results but they were loaded with normal values not zeros
* Also, I have done post-synthesis simulation to check the problem isn't with the elaboration step but with no change

m7md5303, 2025-08-20:
The problem was with the ONNX file input to FINN flow, I was using false one. Nothing related FINN or Vivado

closed 2025-08-20 by m7md5303

auphelia, 2025-08-21:
Hi @m7md5303,

Thank you for updating and closing the PR! Glad to hear that you found the source of the issue.