Xilinx/finn #1430: Check that either folding config or target fps is set

LeahLS, 2025-09-22:
## Prerequisites
None

## Details
### New behavior
There should be an error message early in the build flow if neither a folding config nor the target fps are specified. Therefore, add checks to ensure that either a folding config or a target fps is specified in the building config.

### Motivation
When starting the build flow with neither a folding config nor a target fps set, an error at the hw ipgen step occurs saying that the name of the projects hasn't been set.

### Parts of FINN affected
Transformation, FPGA Dataflow, Set Folding

mumallaeng, 2026-08-24:
Opened #1665, which adds a folding_missing check to run_all_config_checks so this fails fast with a clear message at config-check time, instead of surfacing later as an unrelated error in step_hw_ipgen.

mumallaeng, 2026-08-24:
I traced this through SpecializeLayers/PrepareIP with a minimal PE=1/SIMD=1 MVAU model and it completes cleanly - neither step raises on its own. So the failure mode isn't actually a crash in the current codebase; convert_to_hw_layers already sets PE=1/SIMD=1 at node creation time regardless of folding config. The real risk is that the build silently runs to completion with the most folded, slowest implementation, and that only becomes obvious after a full synthesis run. Updated #1665 to check and message for that instead.

auphelia, 2026-09-16:
Thanks @LeahLS for flagging this issue and @mumallaeng for addressing it, the PR is merged now, so I will close the issue.

closed 2026-09-16 by auphelia